OpenDriveLab/UniVLA #2: During the action decoder post training stage, do you tried to freeze the VLM and only trian the action decoder?

HeegerGao, 2025-05-14:
Thanks again for your great work. I see that in some other works (LAPA, IGOR) that also use lantent action tokens for manipulation tasks, they seem to freeze the VLM (i.e., the latent action encoder) during the action decoder post training stage, but in UniVLA I saw that you fine-tuned the full model (with LoRA). Since the VLM (in the future) may become a very large-scale pretrained model in the embodied AI community, fine-tuning it for specific tasks may not be affordable for everyone.

So do you think it is necessary to also fine-tune the VLM during the post-training stage? Do you have any experimental comparisons on this? I’d like to hear your opinion on this point. Thanks!

retsuh-bqw, 2025-05-14:
Thank you for raising this point. I'd like to first calify that, in LAPA, they only freeze the vision encoder but 'unfreeze all of the parameters of the underlying language model' (I quote), which leads to much more trainable parameters than using LoRA. For IGOR, they didn't tell much about implementation details and downstream policy learning was not thoroughly studied, but they were not using VLMs.

I believe further fine-tuning the VLM part (whether using LoRA or full fine-tuning) is becoming common practise now as you may see in openvla, openvla-oft and pi0, etc. In our preliminary study, LoRA fine-tuning would definitly be helpful. However, if you try finetuning our pre-trained policy on LIBERO (which is unseen from its pre-training dataset), UniVLA's initial latent action accuracy would be around 0.4-0.5, instead of 0. So UniVLA can do, to some point, zero-shot generalization to unseen environments / setups. But given our action decoder is extremely small (~12M params. v.s. 130+M used in OpenVLA-OFT), only trainining the action decoder with other part completly frozen will not give you a satisfactory result. 

Thank you again for raising this question. We are planning to delve deeper into this and update relevant results. We are also considering building UniVLA with more powerful but smaller VLMs to reduce possible training cost.

HeegerGao, 2025-05-14:
Thanks very much for your answer!

closed 2025-05-14 by HeegerGao