OpenDriveLab/UniVLA #4: Can u please share dataset or at least how does latent action model being trained?

lucasjinreal, 2025-05-16:
How to label these data to train a latent action model?
also, the input resolution is 224x224, for manipulation tasks this could be fine, but for navigation, this input size isn't too small to identify any small objects?

retsuh-bqw, 2025-05-16:
Our latent action learning process is inherently unsupervised, requiring only videos and language data (in UniVLA) that are readily available in most existing datasets. So we do not need to 'label' them. All our leveraged datasets (specifically, OpenX and Ego4D) are publicly accessible. 

Regarding resolution: Most established vision-language-navigation models built upon VLMs (e.g., [Uni-NaVid](https://arxiv.org/abs/2412.06224), [NaVid](https://arxiv.org/abs/2402.15852), and [NavGPT](https://arxiv.org/abs/2305.16986), etc) use 224×224 resolution, as their vision backbones are pre-trained at this scale. This resolution would be sufficient for most navigation tasks.

lucasjinreal, 2025-05-17:
For navigation, how's the label organised, is it by text label: `yaw speed, velocity` or something?

lucasjinreal, 2025-05-20:
Thanks for the reply. Still want to consult 2 questions:

1) Currently only ego4d dataset is available for the navi task, which does not fully cover more diverse scenarios. Is there a way to construct video and instruction from webvideos? Any thoughts on this?
2) For videos like 15fps, what frame frequency should be sampled from the raw?

closed 2025-05-25 by retsuh-bqw