OpenDriveLab/UniVLA #5: Stage 2 for latent action token learning

trust-x-embodiment, 2025-05-17:
Hi, thank you for open-sourcing this impressive work!

I'm currently studying your paper and had a question regarding the second stage of latent action learning — specifically, the task-centric latent action decoupling process.

In Stage 2, are only the learnable task-centric latent action tokens and their associated VQ codebook updated, while the encoder and decoder remain frozen from Stage 1? Or are the encoder and decoder also fine-tuned, given that the learning objective shifts and both the encoder output and decoder input structures change in this stage?

I'm asking because freezing the encoder/decoder would align more intuitively with the goal of decoupling task-relevant and task-irrelevant dynamics, but I could also imagine that fine-tuning may be necessary to adapt to the new structure.

Thanks again for sharing this work — I’d really appreciate your clarification!

Best regards

retsuh-bqw, 2025-05-20:
Thank you for your interest in our work!

In Stage 2, only the VQ codebook for task-irrelevant information remains frozen (ensuring it provides consistent input to the decoder as learned in Stage 1), while all other components are trained to optimize performance.

We conducted preliminary studies on training strategies and found that freezing the encoder/decoder leads to unstable training. This instability likely stems from the input distribution shift between the pretrained language embeddings and the newly initialized latent action tokens.

trust-x-embodiment, 2025-05-20:
Hi! Understood. Thank you!

closed 2025-05-20 by trust-x-embodiment

brunomachado37, 2025-05-27:
Hi,

Thank you for the great open-source work !

I have a few questions concerning the stage 2 latent action model training.

Given that your encoder and decoder are unfreeze during this stage while the codebook for task-irrelevant latent actions is frozen, doesn't this ruins the alignment between the learned latent actions and images? For example, an image pair which had a specific task-irrelevant latent action by the end of stage 1 can have a different mapping during stage 2 (the encoder weights are updated while the codebook can't adapt, the latent action can be mapped to a different code).

Also, about the ablation on task-centric and task-irrelevant pre-training. If I understood correctly, the task-irrelevant used here come from the model after the ending of stage 2. Did you try using the model from stage 1 instead ? And did you try predicting both task-centric and task-irrevelant (full output of stage 2) ?

retsuh-bqw, 2025-05-28:
@brunomachado37 Thanks for your interest in our work! Your questions are very insightful.

1. At the start of Stage-2 training, the alignment between learned latent actions and images is inevitably disrupted because the input distribution shifts due to newly initialized action tokens. On the decoder side, we freeze the task-irrelevant latent actions to preserve their learned information, while forcing the newly initialized task-centric latent actions to learn something new (not provided by the task-irrelevant ones). Additionally, adapting the codebook would make it difficult to explicitly decouple features learned by the two groups of latent actions—since they share the same dimension and are treated equally during Stage-2 training. In practice, we empirically found training the encoder and decoder is helpful.

2. Yes, the ablation study was conducted using task-irrelevant latent actions from Stage-2. We haven’t yet performed the study you suggested (due to the lengthy evaluation pipeline involving subsequent pre-training, post-training then closed-loop evaluation). However, we are planning to explore this and update the results to provide further insights into latent action learning.