OpenDriveLab/UniVLA #6: Why evalution process so low?

zhangSaneC, 2025-05-19:
Thanks for your great work!

I think the evaluation process does not utilize the GPU, and running the evaluation with trials=50 takes a very long time.

Does anyone have any ideas on how to speed up the evaluation?

============================================
Task: put both the cream cheese box and the butter in the basket████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▉                                                                                       | 36/50 [34:46<13:27, 57.64s/it]
Starting episode 37...
Success: True
# episodes completed so far: 87
# successes: 84 (96.6%)

Task: put both the cream cheese box and the butter in the basket███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▏                                                                                | 37/50 [35:43<12:27, 57.54s/it]
Starting episode 38...
Success: True
# episodes completed so far: 88
# successes: 85 (96.6%)

bailvwangzi, 2025-05-19:
+1. 

openvla or openvla-oft not so slow.

retsuh-bqw, 2025-05-20:
Thank you for your feedback.

The slower evaluation speed compared to OpenVLA-OFT is due to our implementation of action chunking with temporal integration. By default, we run **inference at every single step** and aggregate action trajectories temporally, similar to [ACT](https://tonyzhaozh.github.io/aloha/).

To improve evaluation speed, we also provide a blocking control script ([run_libero_eval_blocking.py](https://github.com/OpenDriveLab/UniVLA/blob/ea5a973465b1b59236cbd0040f31db1adc8f2164/experiments/robot/libero/run_libero_eval_blocking.py)), where all actions in a chunk are executed before the next model inference (matching OpenVLA-OFT's approach). Theoretically, this can speed up evaluation by nearly `action_chunk_size=12` times.

However, please note that, since UniVLA uses only the one-step prior latent action when learning from history, this blocking method may slightly reduce performance, as it is not compatible with history latent action input—particularly on LIBERO-Long, where historical context is more beneficial.

Additionally, I’d like to clarify that UniVLA’s inference is inherently faster than OpenVLA’s, as UniVLA generates 4 tokens per action chunk, while OpenVLA generates 7 tokens per single-step action.

bailvwangzi, 2025-05-20:
Thank you. Can you provide the LIBERO results of blocking control version?

retsuh-bqw, 2025-05-20:
I haven’t fully tested this across all suites, but generally, on LIBERO-Spatial, -Object, and -Goal, the performance impact is negligible. However, for LIBERO-Long, performance may decrease by 3-4% due to the absence of historical latent action input.

closed 2025-05-22 by retsuh-bqw

Leo-Yuyang, 2025-06-03:
hello, I'm trying to resolve this issue: https://github.com/OpenDriveLab/UniVLA/issues/16
It would be very nice if you can share some clues because you evaluated it successfully! Sorry for bothering you!