OpenDriveLab/UniVLA #7: libero results not good

bailvwangzi, 2025-05-19:
Fineture libero-spatial , the train_loss can converge to 0.06, latent_action_accuracy=1, but when test with run_libero_eval.py the SR < 5%. Do you have any suggestions?

One more thing, there are some typo bugs, 

- [no slow_output](https://github.com/OpenDriveLab/UniVLA/blob/main/vla-scripts/finetune_libero.py#L81)
- [training_phase or training_Phase](https://github.com/OpenDriveLab/UniVLA/blob/main/vla-scripts/finetune_libero.py#L283)
- [calvin_dataset not ready but import now](https://github.com/OpenDriveLab/UniVLA/blob/main/prismatic/vla/datasets/__init__.py#L4)

swstbecrpgmail, 2025-05-20:
[image]
[image]
Have you encountered the following error when running "torchrun --standalone --nnodes 1 --nproc-per-node 3 finetune_libero.py --dataset_name 'libero_10_no_noops' --run_root_dir 'libero_log"? My devices are three A6000 GPUs.
Thank you!

retsuh-bqw, 2025-05-20:
@bailvwangzi Thanks a lot for testing our code and report those issues! I've fixed them in the latest commit.

Regarding the reproduction issue on LIBERO-spatial, I would first recommend verifying the dataset you're using. When training on all LIBERO suites, UniVLA's latent action accuracy typically converges around 0.7–0.8 and is unlikely to reach 1.0. Additionally, the train_loss usually stays within the range of 0.3–0.4.

bailvwangzi, 2025-05-20:
I check again, i use the correct dataset of LIBERO. [This issue](https://github.com/OpenDriveLab/UniVLA/issues/1#issuecomment-2872749587) you said train_loss converges to below 0.04 (i think the answer above you mean 0.03-0.04, not 0.3-0.4). Do you have any other suggestions? The wandb visualization: 

[image]

retsuh-bqw, 2025-05-20:
The training curve seems quite unusual. Could you please provide detailed training setups? Also, are you using the correct latent action model and pretrained UniVLA checkpoints we provided?

My training curve on libero-spatial is like (train_loss = action_loss + next_latent_action_token_prediction_loss):

[image]

bailvwangzi, 2025-05-20:
[image]

I use 3090 GPU * 2, batchsize=1, with lower learning_rate and large max_steps.

bailvwangzi, 2025-05-21:
> [image]
> 
> I use 3090 GPU * 2, batchsize=1, with lower learning_rate and large max_steps.

All the changes are shown in the picture above. The vla_path ,lam_path, data_root_dir are right? @retsuh-bqw

retsuh-bqw, 2025-05-21:
The current batch size (GPUs $\times$ batch_size $\times$  grad_accumulation_steps = 4) is significantly smaller than our experimental setting (128). To address this, you could either: 1. Simply use more GPUs, or 2. Increase the `grad_accumulation_steps` to achieve a larger effective batch size.

bailvwangzi, 2025-05-21:
> The current batch size (GPUs × batch_size × grad_accumulation_steps = 4) is significantly smaller than our experimental setting (128). To address this, you could either: 1. Simply use more GPUs, or 2. Increase the `grad_accumulation_steps` to achieve a larger effective batch size.

yes, i also try to use large grad_accumulation_steps=32, but  curve also not same with you.

[image]

retsuh-bqw, 2025-05-22:
I replicated your training configuration but noticed that the latent action accuracy exhibits significant fluctuations rather than converging quickly to 1.0. I also verified the integrity of the uploaded checkpoints via checksum, confirming they are correct. I've also tried to reproduce your results with 1 GPU but unfortunately failed.

[image]

closed 2025-05-23 by retsuh-bqw

bailvwangzi, 2025-05-23:
The error loss curve is because i use wrong **transformers** version. `pip install transformers==4.40.1` is right. Thanks again.

Keyustar, 2025-06-16:
So it's normal that a 24GB 4090 can't handle DDP training with the default configuration, is that correct? I tried replacing DDP with DeepSpeed for training, and it works under Zero Stage 2, but the training results are not ideal. I used batch_size=1, grad_accumulation_steps=1, 8 GPUs, and max_steps=60,000—is this due to the effective batch size being too small?

> > [@bailvwangzi](https://github.com/bailvwangzi) Hello, may I ask how much VRAM your GPU has? Can it perform training under default settings without Out Of memery? I'm using 8 * 4090 GPUs, 和 it always encounters OOM with the default settings.
> 
> The full model is loaded onto each GPU. I use RTX5880 ada with batch_size=1, which consumes approximately 32GB of VRAM.

So it's normal that a 24GB 4090 can't handle DDP training with the default configuration, is that correct? I tried replacing DDP with DeepSpeed for training, and it works under Zero Stage 2, but the training results are not ideal. I used batch_size=1, grad_accumulation_steps=1, 8 GPUs, and max_steps=60,000—is this due to the effective batch size being too small?