OpenDriveLab/UniVLA #12: get sucess rate =0 when eval on libero

Luo-Zhongwei, 2025-05-23:
here are my log results and  bash 

```python
512
2025-05-22 21:19:57
Starting episode 48...
3513
2025-05-22 21:20:49
Saved rollout MP4 at path ./rollouts/2025_05_22/2025_05_22-21_45_55--episode=498--success=False--task=put_the_wine_bottle_on_the_rack.mp4
3514
2025-05-22 21:20:49
Success: False
3515
2025-05-22 21:20:49
# episodes completed so far: 498
3516
2025-05-22 21:20:49
# successes: 0 (0.0%)
3517
2025-05-22 21:20:49

3518
2025-05-22 21:20:49
Task: put the wine bottle on the rack
3519
2025-05-22 21:20:51
Starting episode 49...
3520
2025-05-22 21:21:44
Saved rollout MP4 at path ./rollouts/2025_05_22/2025_05_22-21_45_55--episode=499--success=False--task=put_the_wine_bottle_on_the_rack.mp4
3521
2025-05-22 21:21:44
Success: False
3522
2025-05-22 21:21:44
# episodes completed so far: 499
3523
2025-05-22 21:21:44
# successes: 0 (0.0%)
3524
2025-05-22 21:21:44

3525
2025-05-22 21:21:44
Task: put the wine bottle on the rack
3526
2025-05-22 21:21:46
Starting episode 50...
3527
2025-05-22 21:22:38
Saved rollout MP4 at path ./rollouts/2025_05_22/2025_05_22-21_45_55--episode=500--success=False--task=put_the_wine_bottle_on_the_rack.mp4
3528
2025-05-22 21:22:38
Success: False
3529
2025-05-22 21:22:38
# episodes completed so far: 500
3530
2025-05-22 21:22:38
# successes: 0 (0.0%)
3531
2025-05-22 21:22:38
Current task success rate: 0.0
3532
2025-05-22 21:22:38
Current total success rate: 0.0

```

```python
3525
2025-05-22 21:06:10
Task: pick up the orange juice and place it in the basket
3526
2025-05-22 21:06:12
Starting episode 50...
3527
2025-05-22 21:07:04
Saved rollout MP4 at path ./rollouts/2025_05_22/2025_05_22-21_43_23--episode=500--success=False--task=pick_up_the_orange_juice_and_place_it_in_the_baske.mp4
3528
2025-05-22 21:07:04
Success: False
3529
2025-05-22 21:07:04
# episodes completed so far: 500
3530
2025-05-22 21:07:04
# successes: 0 (0.0%)
3531
2025-05-22 21:07:04
Current task success rate: 0.0
3532
2025-05-22 21:07:04
Current total success rate: 0.0
```

```python
3020
2025-05-23 02:24:42

3021
2025-05-23 02:24:42
Task: put the yellow and white mug in the microwave and close it
3022
2025-05-23 02:24:44
Starting episode 49...
3023
2025-05-23 02:26:14
Success: False
3024
2025-05-23 02:26:14
# episodes completed so far: 499
3025
2025-05-23 02:26:14
# successes: 0 (0.0%)
3026
2025-05-23 02:26:14

3027
2025-05-23 02:26:14
Task: put the yellow and white mug in the microwave and close it
3028
2025-05-23 02:26:16
Starting episode 50...
3029
2025-05-23 02:27:47
Success: False
3030
2025-05-23 02:27:47
# episodes completed so far: 500
3031
2025-05-23 02:27:47
# successes: 0 (0.0%)
3032
2025-05-23 02:27:47
Current task success rate: 0.0
3033
2025-05-23 02:27:47
Current total success rate: 0.0

```

Luo-Zhongwei, 2025-05-23:
here are python bash 

```python
CUDA_VISIBLE_DEVICES=0 \
python experiments/robot/libero/run_libero_eval.py \
    --task_suite_name libero_spatial  \
    --action_decoder_path /root/autodl-tmp/ckpt/univla_ckpt/univla-latent-action-model/lam-stage-2.ckpt\
    --pretrained_checkpoint /root/autodl-tmp/ckpt/univla_ckpt/univla-7b-224-sft-libero/univla-libero-spatial \
    --save_video True \
    --seed 7 \
    --use_wandb True \
    --wandb_project univla_libero_eval \
    --wandb_entity  485434977-shang-hai-jiao-tong

CUDA_VISIBLE_DEVICES=1 \
python experiments/robot/libero/run_libero_eval.py \
    --task_suite_name libero_goal  \
    --action_decoder_path /root/autodl-tmp/ckpt/univla_ckpt/univla-latent-action-model/lam-stage-2.ckpt\
    --pretrained_checkpoint /root/autodl-tmp/ckpt/univla_ckpt/univla-7b-224-sft-libero/univla-libero-goal \
    --save_video True \
    --seed 7 \
--use_wandb True \
    --wandb_project univla_libero_eval \
    --wandb_entity  485434977-shang-hai-jiao-tong

CUDA_VISIBLE_DEVICES=2 \
python experiments/robot/libero/run_libero_eval.py \
    --task_suite_name libero_object  \
    --action_decoder_path /root/autodl-tmp/ckpt/univla_ckpt/univla-latent-action-model/lam-stage-2.ckpt\
    --pretrained_checkpoint /root/autodl-tmp/ckpt/univla_ckpt/univla-7b-224-sft-libero/univla-libero-object \
    --save_video True \
    --seed 7 \
--use_wandb True \
    --wandb_project univla_libero_eval \
    --wandb_entity  485434977-shang-hai-jiao-tong

CUDA_VISIBLE_DEVICES=0 \
python experiments/robot/libero/run_libero_eval.py \
    --task_suite_name libero_10  \
    --action_decoder_path /root/autodl-tmp/ckpt/univla_ckpt/univla-latent-action-model/lam-stage-2.ckpt\
    --pretrained_checkpoint /root/autodl-tmp/ckpt/univla_ckpt/univla-7b-224-sft-libero/univla-libero-10 \
    --save_video False \
    --seed 7 \
--use_wandb True \
    --wandb_project univla_libero_eval \
    --wandb_entity  485434977-shang-hai-jiao-tong

```

retsuh-bqw, 2025-05-23:
The evaluation should work fine (also see this [issue](https://github.com/OpenDriveLab/UniVLA/issues/6#issue-3072612604) for reference). The correct `action_decoder_path` should be **`/root/autodl-tmp/ckpt/univla_ckpt/univla-7b-224-sft-libero/univla-libero-10/action_decoder.pt`** (for libero-10), not the latent action model. I am also curious though - how were you able to run the evaluation without encountering errors when loading the wrong checkpoint?

Luo-Zhongwei, 2025-05-24:
> /root/autodl-tmp/ckpt/univla_ckpt/univla-7b-224-sft-libero/univla-libero-10/action_decoder.pt

what 's more., so the correct  path is /root/autodl-tmp/ckpt/univla_ckpt/univla-7b-224-sft-libero/univla-libero-10/action_decoder.pt

so what is the usage of   /root/autodl-tmp/ckpt/univla_ckpt/univla-latent-action-model/lam-stage-2.ckpt

retsuh-bqw, 2025-05-24:
This is our latent action model, which generates pseudo-labels (predicting latent actions from current and future frames) to train our generalist policy. For more details, please refer to our paper.

Leo-Yuyang, 2025-06-03:
hello, I'm trying to resolve this issue: https://github.com/OpenDriveLab/UniVLA/issues/16
It would be very nice if you can share some clues because you evaluated it successfully! Sorry for bothering you!