OpenDriveLab/UniVLA #13: Online inference script of LAM model

lucasjinreal, 2025-05-25:
Hi, I'm wondering:

1) is there a script to infer the LAM model only with real-world videos? I want to see how it behaves for a simple task like navigation;
2) How to use the existing LAM to easily train a Qwen2.5 VL based two - system model? Currently it heavily relies on primisma which is a little complicated compared with SOTA vlms.

retsuh-bqw, 2025-06-03:
LAM does not need to resolve or interpolate them. LAM is only to generate pseudo labels for VLM pretraining. I believe your are referring to the action deocder, which is built upon a multi-head attention pooling module and a MLP to directly decode latent action embeddings (aka, the VLM's last-layer faeture corresponding to latent action tokens, not the decoded <ACT_i> tokens in the form of text) into low-level action chunks with the L1 regression loss. So LAM is NOT used to decode the latent actions and it is completly deprecated during inference time.

lucasjinreal, 2025-06-03:
hi, does the navigation task is accurate and robust? what if target is missing a long time, will it resume instructions?