OpenDriveLab/UniVLA #14: Question about LAM stage 2 didn't use text encoder

lucasjinreal, 2025-05-25:
I know there are some instructions in the paper that in stage1 the tasks are already decoupled, but the question is, why? Why doesn't LAM stage 2 have a text encoder? What if I want to do some further complicated tasks in stage2 specifically in the Nav task, for example, I need to specify the go - to target in the nav task?

Is there any further reason why stage 2 didn't have a text input condition?

fwd4, 2025-06-13:
I'm very curious about this issue as well. 
- The 1st stage is called **task-irrelevant** while task instructions are provided. 
- The 2nd stage is called **task-centric** while task information is not given.
What are the reasons behind those terminologies?

retsuh-bqw, 2025-06-13:
Thanks for bringing this up! During the first stage of training, the strong information bottleneck created by the VQ-VAE codebook forces the learned latent actions to capture only the information not inferable from task instructions—resulting in task-irrelevant latent actions.

In the second stage, with these task-irrelevant latents frozen, the newly initialized latent actions are trained to replace the role of language, allowing them to learn task-centric dynamics.

This also explains why, in naive latent action learning approaches like LAPA and Genie, latent action tokens mostly encode motion/dynamics—since texture information is already preserved from previous frames for future frame prediction in the decoder.

dreamcubeblock, 2025-06-23:
If your goal is to learn task-relevant tokens, why not directly use contrastive learning to encode those tokens and perform a soft alignment with the text instruction features? Have you tried this approach? It seems like a more straightforward way to achieve your objective.

JisuHann, 2025-06-30:
@retsuh-bqw Thanks for your detailed explanation.

To support your claims, I think it would be helpful to demonstrate what kind of behavior the latent actions actually exhibit. Have you tried comparing them in a way similar to LAPA Figure 13? Or perhaps visualizing the reconstructions like in LAPA Figure 14?

retsuh-bqw, 2025-07-03:
> If your goal is to learn task-relevant tokens, why not directly use contrastive learning to encode those tokens and perform a soft alignment with the text instruction features? Have you tried this approach? It seems like a more straightforward way to achieve your objective.

@dreamcubeblock We haven't tried this approach yet, but here's our reasoning: During a task episode, the latent action should evolve over time while the language instruction remains the same. Explicit contrastive alignment could harm convergence in this case, as it may be more suitable for learning latent actions that represent the entire task rollout rather than individual steps.

retsuh-bqw, 2025-07-03:
> [@retsuh-bqw](https://github.com/retsuh-bqw) Thanks for your detailed explanation.
>
> To support your claims, I think it would be helpful to demonstrate what kind of behavior the latent actions [...]

@JisuHann Thank you for your suggestion. While we cannot perform explicit reconstruction (since LAM operates in DINO space rather than raw pixels), we have conducted related analyses:

- Figures 8 & A-1 demonstrate that latent actions exhibit semantic consistency across environments and embodiments.

- Figure A-3 explores the attention patterns between latent actions and DINO features.