OpenDriveLab/UniVLA #20: Question about load latent action model stage1 checkpoint

luzhoulz, 2025-06-12:
in UniVLA/latent_action_model/genie/model.py , line 67 - line 73:
```
        if stage_one_ckpt and path.exists(stage_one_ckpt):
            lam_ckpt = torch.load(stage_one_ckpt)['state_dict']
            stage1_ckpt = {}
            for key in lam_ckpt.keys():
                if 'vq' in key or 'action_latent' in key:
                    stage1_ckpt[key.replace("lam.", "")] = lam_ckpt[key]
            self.lam.load_state_dict(stage1_ckpt, strict=False)
```

Why only parameters whose name contains `vq` or `action_latent` are loaded? Other parameters such as `encoder*` `decoder*` `to_codebook` `action_up` from stage1 are discarded?

retsuh-bqw, 2025-06-17:
We've simply tried several strategies, and empirically found it works best (based on downsteam policy performance) and aligns with our goals.

luzhoulz, 2025-06-18:
> We've simply tried several strategies, and empirically found it works best (based on downsteam policy performance) and aligns with our goals.

Thanks for your explanation

closed 2025-06-18 by luzhoulz