OpenDriveLab/UniVLA #25: Train time

DeepDarkDuck, 2025-06-22:
Thanks for open-sourcing this great work! I saw that the time required in "Pre-Training" on the 'Bridge' subset is 200 GPU-hours. But I find that I need 800 hours to pre-train with 8 A100 GPU. So I want to know the meaning of "200 GPU-hours": "32 A100-->200h" or "32 A100-->200h/32"?

[image]

retsuh-bqw, 2025-06-22:
The training script will run for 50 epoches by default. But we've actually used the checkpoint at 20,000 optimization steps (approximately 5 epoches of training). '200 GPU-hours' means we trained on a 8 $\times$ A100 node for about one day.

DuckSQ-BIT, 2025-06-22:
> The training script will run for 50 epoches by default. But we've actually used the checkpoint at 20,000 optimization steps (approximately 5 epoches of training). '200 GPU-hours' means we trained on a 8 × A100 node for about one day.

Thanks for your reply. But I find that the epoches are set as 10 by default actually. And I just change "data_root_dir" before training and find that we need 800 hours only using bridge subset with 8 A100 GPU. So I want to know how to config when training on bridge subset.

[image]

retsuh-bqw, 2025-06-23:
If you're using `bridge_orig`, note that 20,000 optimization steps only equals about 2 epochs. The max_steps parameter might be calculated to be much larger than necessary, leading to an unexpectedly long training time. However, you only need to use the 20,000-step checkpoint anyway.

For reference, here's a screenshot of the training log from my setup with 8 A100 GPUs:
[image]

DuckSQ-BIT, 2025-06-23:
> If you're using `bridge_orig`, note that 20,000 optimization steps only equals about 2 epochs. The max_steps parameter might be calculated to be much larger than necessary, leading to an unexpectedly long training time. However, you only need to use the 20,000-step checkpoint anyway.
> 
> For reference, here's a screenshot of the training log from my setup with 8 A100 GPUs: <img alt="Image" width="1040" src="https://private-user-images.githubusercontent.com/64821056/457725881-5be7ca63-9872-4339-a3c7-8f90f23ca23a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTA2NzA3NDksIm5iZiI6MTc1MDY3MDQ0OSwicGF0aCI6Ii82NDgyMTA1Ni80NTc3MjU4ODEtNWJlN2NhNjMtOTg3Mi00MzM5LWEzYzctOGY5MGYyM2NhMjNhLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA2MjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwNjIzVDA5MjA0OVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUxM2I5NDZjZjY3N2FjMzc3YzYzOGQ4OWRjZTFiNjY4YjlmN2I3NmViYTFhMGVkZGZlMTQ1MWQ3YzU0NDcxOWQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.nT_T6evjDXW68-j1auugQ-5ACQBpeztTYyuNa5TFIaI">

Thanks for your reply! And I want to know if the training of latent action tokenizer has similar problem? When we trained latent action tokenizer on stage-1 only using bridge subset with 8 A100 GPU, we found that it needed 9h per epoch.

[image]

retsuh-bqw, 2025-07-03:
For our release LAM trained collectively on 'Manipulation + Navigation + Human', we trained stage-1 for 150k steps and stage-2 for 200k steps. Given the full dataset size, 200k steps corresponds to roughly 1 epoch of training. 

You may consider early stopping once the loss clearly stabilizes during training on custom datasets.