OpenDriveLab/UniVLA #26: Can't reporduce the results on libero_spatial

dreamcubeblock, 2025-06-23:
We fine-tuned the model on an A100 GPU with the same basic hyperparameters, except that we modified shuffle_buffer_size to prevent OOM.  In the final evaluation, the accuracy was less than 60%. Could you give me some advice?  Thank you!

[image]

[image]

retsuh-bqw, 2025-06-23:
I noticed that you're training on a single GPU, which means your effective batch size is only 1/8 of our original setup while using the same large learning rate. To better align with our training configuration, you might consider:

1. Using multiple GPUs to match our setup (if resources allow), or

2. Reducing the learning rate (e.g., to ~5e-5) and increasing the number of optimization steps to compensate.

dreamcubeblock, 2025-06-23:
Thank you so much, I will try it right now!