OpenDriveLab/UniVLA #32: How to split Ego4d dataset?

DuckSQ-BIT, 2025-07-03:
Hi. Thanks for open-sourcing this great work! I have a question about the segmentation of Ego4d datasets. In "UniVLA/prismatic/vla/datasets/rlds/oxe/mixtures.py" I found that Ego4d was splited as "ego4d_split_x"(x=1-4). I want to know how you split the dataset? Moreover, it seems that data is loaded by "builder = tfds.builder(name, data_dir=data_dir)", meaning that Ego4d dataset needs to be converted to “tfds” format. Can you tell me how to convert Ego4D dataset to tfds-format?

[image]

JackHuang0701, 2025-07-14:
Hi @DuckSQ-BIT , have you solved this question?

retsuh-bqw, 2025-07-21:
I've updated the instruction for processing Ego4D-HOI dataset [here](https://github.com/OpenDriveLab/UniVLA/blob/2b780a5992406262680c6c66150564b3d8befaf2/vla-scripts/extern/ego4d_rlds_dataset_builder/ego4d/README.md). Please reach out further if you have any questions.

closed 2025-07-21 by retsuh-bqw

DuckSQ-BIT, 2025-07-22:
> I've updated the instruction for processing Ego4D-HOI dataset [here](https://github.com/OpenDriveLab/UniVLA/blob/2b780a5992406262680c6c66150564b3d8befaf2/vla-scripts/extern/ego4d_rlds_dataset_builder/ego4d/README.md). Please reach out further if you have any questions.

Thanks for your reply! I'll try your code to process ego4d dataset. If I face other problems, I'll connect with you for help or share my solution. I am extremely grateful for your earnest and consistent replies!