OpenDriveLab/UniVLA #46: ego4d data processing script

NNNatsuki, 2025-07-28:
Thank you for sharing your excellent work.
When processing the ego4d data as you described, the `annotations.json` generated by `preprocess_ego4d.py` in step 2 does not contain the `'id'` key, which causes `create_episode_ego4d.py` in step 3 to not run as expected. Could you please provide some solutions or suggestions?

retsuh-bqw, 2025-07-30:
Thanks for testing our codes. I've fixed this in the latest commit.

NNNatsuki, 2025-07-31:
> Thanks for testing our codes. I've fixed this in the latest commit.

After processed by step3, we only get 812 `episode_.npy` files. This seems to be inconsistent with `Readme`.

JackHuang0701, 2025-08-03:
After processed by step3, I only get 427 episode_.npy files. @retsuh-bqw

JackHuang0701, 2025-08-03:
@retsuh-bqw @hli2020

Nemo-1024, 2025-08-13:
This problem arises from overwriting between different segments. Each segment is named by the selected video frame ID after conversion.

Change
`save_path = os.path.join(save_dir, f'episode_{episode_id}.npy')` 
as
`save_path = os.path.join(save_dir, f'episode_{video_name}_{episode_id}.npy')`

However, you may encounter other errors when executing this pipeline. My response is only for data volume anomalies.

closed 2025-08-13 by NNNatsuki