OpenDriveLab/UniVLA #60: Loss instability when resuming pretraining from converted HF weights

minghaoguo20, 2025-08-20:
Thanks for your brilliant work.

I am currently trying to run **Stage 2 - Pretraining of the Generalist Policy** on the **Bridge V2 dataset**. Since the weights of univla-7b-bridge-pt in UniVLA’s Prismatic format were not released (only the Hugging Face format was made available), I implemented a script convert_hf_to_univla_weights.py (referring to vla-scripts/extern/convert_openvla_weights_to_hf.py) to convert the released HF weights back into UniVLA Prismatic format.

Using the converted weights, I resumed training with settings `--resume_step 0 --resume_epoch 0`.

Theoretically, **since the weights are already pretrained, I would expect the training loss to start from a well-converged state**.

However, **I observed that the loss is quite unstable**. After around 3000 steps, **most of the loss values lie in the 0.5–2.0 range, but many steps even exceed 3**.

I would like to confirm:
	1.	Is this kind of loss behavior normal in your experience?
	2.	When you resumed from the official UniVLA Prismatic weights with resume_step / resume_epoch specified, what range of loss values did you typically observe?

Any clarification would be greatly appreciated.

[image]

感谢你们在 UniVLA 上的出色工作。

我目前正在尝试在 Bridge V2 数据集上运行 Stage 2 - Pretraining of the Generalist Policy。由于你们没有公开 univla-7b-bridge-pt 的 Prismatic 格式权重（只提供了 Hugging Face 格式），我参考了 vla-scripts/extern/convert_openvla_weights_to_hf.py，实现了 convert_hf_to_univla_weights.py，将公开的 HF 权重转回到 UniVLA 的 Prismatic 格式。

基于转换得到的权重，我使用参数`--resume_step 0 --resume_epoch 0`继续训练。

理论上，既然权重已经是预训练过的，一开始的 loss 应该处于比较低的收敛状态。

然而，我发现 loss 非常不稳定。训练至大约 3000 steps 后，大部分 loss 落在 0.5–2.0 区间，但有不少 step 会超过 3。

因此想请教：
	1.	这种剧烈的 loss 波动是否正常？
	2.	当你们基于官方 UniVLA Prismatic 权重并设置了 resume_step / resume_epoch 恢复训练时，loss 大致处于什么范围？

非常感谢你们的解答！

retsuh-bqw, 2025-08-21:
Thanks for reporting this. The losses you're seeing are indeed much higher than expected. To help us investigate, could you please confirm if you're using the OXE-version of the BridgeV2 data? One other possibility is that there might be an issue in the weights conversion script. Let's rule that out—could you double-check that process on your end? 

I'd like to show the loss curve of our pre-training on BridgeV2 data here for your reference:

[image]

minghaoguo20, 2025-08-21:
Hi, I just verified the conversion process and found that that model converted twice (from hf model to univla model to hf model) are the same with the model your team released. I suppose the model conversion code are correct. I put the code below: `vla-scripts/extern/convert_hf_to_univla_weights.py` and `vla-scripts/extern/compare_hf_model.py`

```python
# vla-scripts/extern/convert_hf_to_univla_weights.py
"""
convert_hf_to_univla_weights.py

Converts HuggingFace format UniVLA models back to the native UniVLA checkpoint format
that can be loaded by train.py's pretrained_checkpoint parameter.
"""

import json
import os
import shutil
from dataclasses import dataclass, field
from pathlib import Path
from typing import Dict, Union
import torch
from transformers import AutoModel, AutoTokenizer
import draccus

@dataclass
class HFToUniVLAConfig:
    # fmt: off
    hf_model_path: Union[str, Path] = (                                 # Path to HF format model
        "/path/to/univla-7b-bridge-pt"
    )
    output_path: Union[str, Path] = (                                   # Output path for UniVLA format
        "/path/to/univla-7b-bridge-pt-native"
    )
    checkpoint_name: str = "converted-checkpoint.pt"                    # Checkpoint filename
    overwrite: bool = False                                             # Overwrite existing files
    # fmt: on

    def __post_init__(self) -> None:
        self.hf_model_path = Path(self.hf_model_path)
        self.output_path = Path(self.output_path)
        
        if self.output_path.exists() and not self.overwrite:
            raise FileExistsError(
                f"Output path `{self.output_path}` already exists! "
                "Set `overwrite=True` to overwrite existing files."
            )

# Reverse key mappings from convert_univla_weights_to_hf.py
REVERSE_PROJECTOR_KEY_MAPPING = {
    "projector.fc1.weight": "projector.0.weight",
    "projector.fc1.bias": "projector.0.bias", 
    "projector.fc2.weight": "projector.2.weight",
    "projector.fc2.bias": "projector.2.bias",
    "projector.fc3.weight": "projector.4.weight", 
    "projector.fc3.bias": "projector.4.bias",
}

def remap_hf_state_dict_to_native(hf_state_dict: Dict[str, torch.Tensor], use_fused_vision_backbone: bool = True) -> Dict[str, Dict[str, torch.Tensor]]:
    """Convert HF state dict back to native UniVLA format."""
    
    vision_backbone_state_dict = {}
    projector_state_dict = {}
    llm_backbone_state_dict = {}
    
    for key, value in hf_state_dict.items():
        if key.startswith("projector."):
            # Map projector keys back to native format
            if key in REVERSE_PROJECTOR_KEY_MAPPING:
                projector_state_dict[REVERSE_PROJECTOR_KEY_MAPPING[key]] = value
        
        elif key.startswith("language_model."):
            # Map language model keys back to native format
            native_key = key.replace("language_model.", "llm.")
[... 429 lines cut ...]

    # Compare models
    results = comparator.compare_models(state_dict1, state_dict2)

    # Print summary
    comparator.print_summary(results)

    # Save detailed results if requested
    if cfg.output:
        cfg.output.parent.mkdir(parents=True, exist_ok=True)

        with open(cfg.output, "w") as f:
            json.dump(results, f, indent=2, default=str)

        print(f"\nDetailed results saved to: {cfg.output}")

    # Exit with error code if models are different
    if not results["models_identical"]:
        exit(1)

if __name__ == "__main__":
    compare_hf_models()
```

Could you plz help check if you can reproduce the result you showed above? Thanks in advance.

minghaoguo20, 2025-08-22:
Hi, I just tried the bridge dataset downloaded from [https://github.com/moojink/rlds_dataset_mod](https://github.com/moojink/rlds_dataset_mod). **The loss is also far from normal.** 

Could you please help check if there is anything wrong? Thanks. 

[image]

Nemo-1024, 2025-11-18:
Hello, could you please share the validation loss curves from your training? I noticed that the VLM struggles to predict latent actions accurately—in my training, the action prediction accuracy on the validation set is quite low. @retsuh-bqw @minghaoguo20

retsuh-bqw, 2025-11-19:
We didn't log any validation losses during training. The latent action accuracy may vary from datasets (e.g. over 90% on LIBERO and around 70% for Bridgev2) . Can you be more specific on your case?