OpenDriveLab/UniVLA #61: Training task-centric codebook (LAM stage-2)

feiyu12138, 2025-08-31:
Hi,

Thanks for inspiring work! I'm trying to reproduce LAM training. For efficiency, I'm training the LAM only on BridgeV2. I train the stage-1 model by 40K steps with MSE loss at 1.8e-5. The abnormal thing I found is MSE loss in stage-2 is much higher than the one stage-1(1e-2 v.s. 5e-4 at 8K step). What's more, "q_loss_uncontrol" and "commit_loss_uncontrol" are also unstable and fluctuate in stage-2. 

Have you tried train LAM in such scale dataset? Do you have any suggestion to debug the loss problem?

Thank you for your kindly suggestion in advance!

retsuh-bqw, 2025-09-02:
Please test the updated latent action model from our latest commit. The key change is that language embeddings are now retrieved from the T5 encoder instead of the LAM encoder.

**Expected Behavior:**

- You should observe a similar loss range in both training stages with this updated version.

- Some instability in the loss for task-irrelevant action tokens (loss_uncontrol) is expected, as these weights are frozen during stage-2 training.

feiyu12138, 2025-09-02:
Thanks for prompt response! After I pull the updated code, the loss of first stage raises higher(which I think falls normal ) and stage-2 training is stable now.

closed 2025-09-02 by feiyu12138